arduino-libraries/Arduino_LSM6DS3 #3: Additional examples

sandeepmistry, 2019-07-25:
Suggested by @tigoe:

> * six faces (you did this for the Curie as I recall)
> * accelerometer tap
> * gyroscope as a knob, rotating around the Z axis

tigoe, 2019-08-04:
Here's an example with the Madgwick library that I tested with the Processing visualizer sketch in the Madgwick folder. It's delivering pretty good results, someone else might want to test it:

````
#include <Arduino_LSM6DS3.h>
#include <MadgwickAHRS.h>

// initialize a Madgwick filter:
Madgwick filter;
// sensor's sample rate is fixed at 104 Hz:
const float sensorRate = 104.00;

void setup() {
  Serial.begin(9600);
  // attempt to start the IMU:
  if (!IMU.begin()) {
    Serial.println("Failed to initialize IMU");
    // stop here if you can't access the IMU:
    while (true);
  }
  // start the filter to run at the sample rate:
  filter.begin(sensorRate);
}

void loop() {
  // values for acceleration & rotation:
  float xAcc, yAcc, zAcc;
  float xGyro, yGyro, zGyro;
  
  // values for orientation:
  float roll, pitch, heading;
  // check if the IMU is ready to read:
  if (IMU.accelerationAvailable() &&
      IMU.gyroscopeAvailable()) {
    // read accelerometer & gyrometer:
    IMU.readAcceleration(xAcc, yAcc, zAcc);
    IMU.readGyroscope(xGyro, yGyro, zGyro);
    
    // update the filter, which computes orientation:
    filter.updateIMU(xGyro, yGyro, zGyro, xAcc, yAcc, zAcc);

    // print the heading, pitch and roll
    roll = filter.getRoll();
    pitch = filter.getPitch();
    heading = filter.getYaw();
    Serial.print("Orientation: ");
    Serial.print(heading);
    Serial.print(" ");
    Serial.print(pitch);
    Serial.print(" ");
    Serial.println(roll);
  }
}
````

tigoe, 2019-08-04:
Updated the madgwick example to use handshaking, and added a p5.js visualizer sketch that works with it:
[arduino sketch with handshaking](https://github.com/ITPNYU/physcomp/blob/master/Labs/LabIMUs/Nano33Madgwick/Nano33Madgwick.ino)
[p5.js visualizer](https://github.com/ITPNYU/physcomp/tree/master/Labs/LabIMUs/MadgwickVisualizer)

tigoe, 2019-08-25:
Here's an update on the p5.js visualizer. Added some [content to explain it](https://itp.nyu.edu/physcomp/labs/lab-serial-imu-output-to-p5-js/).

tkhabia, 2020-02-29:
hello,
I am Tanmay khabia from IIIT Hyderabad. I have worked on various sensors and uses them for various IoT projects.  
I wish to work on this issue as a GSoC project.
what is meant by six faces? 
which direction are we tapping from on the sensor?
thank you

tigoe, 2020-02-29:
Six faces means six possible orientations of the board. See https://www.arduino.cc/en/Tutorial/Genuino101CurieIMUAccelerometerOrientation for an example of it.

tkhabia, 2020-03-07:
please guide me if you want further improvement in the pull request.

MisterAwesome23, 2020-03-31:
Hey @sandeepmistry :)

For task 3- **Gyroscope as a knob, rotating around the Z axis**
Do we need something like- https://www.youtube.com/watch?v=vSKEH0FwhUE

Just instead of controlling a servo via the same, do we need to values on Serial Monitor (and keep things simpler) to get orientations as we move our knob (LSM6DS3).