arduino-libraries/Arduino_LSM6DS3 #5: Built-in FIFO buffer of the Nano 33 IoT

jojo1212, 2019-08-27:
Use of built-in buffer for LSM6DS3 library for nano 33 IoT.

falltec, 2019-08-28:
Hi everybody!
I am doing a project where I want to transmit the accelerometer data at a frequency of about 100 Hz via Wifi to a server. With the "normal" use of the IMU, there are reading interruptions of 50-400ms (depending on the connection strength). This happens when the microcontroller is busy sending the data via HTML-Post request. 
With the Fifo-Buffer, the IMU would be able to save the data for a small amount of time (0.1-1s) and the microcontroller could read it out afterwards. This already worked with the external MPU6050 IMU, but the Nano 33 IoT would be a smaller and better solution.

kaiiorg, 2020-06-18:
I've gotten FIFO usage to work in my quick and dirty proof of concept but the device has a limitation: it doesn't tell you what the data you're actually reading off the FIFO is. After a lot of trial and error, I've gotten to the point where I'm reading the data off and am apparently getting two samples of accelerometer data to one same of gyro data. Not sure what's causing that at the moment.

[image]

kaiiorg, 2020-06-19:
After a lot more poking, prodding, and datasheet reading, I've found that the `FIFO_STATUS3` and `FIFO_STATUS4` registers correspond to the values I'm getting out of the FIFO. I still haven't quite nailed down what rules are used to determine the correspondence, other than dataset order: 1st dataset is the gyro, 2nd is the accelerometer. 

I'm also still not sure why I'm getting two accelerometer readings to one gyroscope reading. And I'm not sure why this data is now in a different order from before. Might have gotten out of sync and everything shifted over.

[image]

kaiiorg, 2020-06-19:
On closer inspection, it turns out I was decimating the gyro data. Off by one bit when configuring the gyro/accelerometer FIFO settings in `LSM6DS3_FIFO_CTRL3`.

Order of columns: 
`gX`  `gY`  `gZ`  `aX`  `aY`  `aZ`
[image]