arduino-libraries/Arduino_LSM6DS3 #7: Possbile error in LSM6DS3.cpp

Alberto-SM, 2020-01-23:
I'm afraid that these lines in LSM6DS3.cpp are not correct:

`93 // Set the ODR config register to ODR/4`
`94 writeRegister(LSM6DS3_CTRL8_XL, 0x09);`

Actually, ODRs are set in CTRL1_XL and CTRL2_G.
I discovered this beacuse that line was making not possible to set up the High Pass filter.

Rocketct, 2020-01-28:
Hi @Alberto-SM i checked this is a legacy of our LSM6DSM library, checking the Figure 6. Accelerometer composite filter, this line have only the effect of enable the LOW_PASS_ON_6D  and setting to 1 a bit that should be set to 0 because reserved, i'll remove because added only for redundancy in the other library and here don't make sense to exist.

per1234, 2022-08-22:
@Rocketct the lines in question are still there:

https://github.com/arduino-libraries/Arduino_LSM6DS3/blob/5ab6cab7c6f45c7496cf64f6a7ff3e46391c79c5/src/LSM6DS3.cpp#L67-L68

I see you merged the pull request which was stated as resolving this, but that pull request contained 0 commits at the time it was merged, so it had no effect on the repository:

https://github.com/arduino-libraries/Arduino_LSM6DS3/pull/8/files