arduino-libraries/Arduino_LSM6DS3 #17: For I2C, end() method calls for _wire->end() instead of _wire->endTransmission()

cjinn, 2020-11-26:
I noticed, when trying out this library for a ESP32 and I2C project, this library does not compile due to the following errors:

```
libraries\Arduino_LSM6DS3\src\LSM6DS3.cpp: In member function 'void LSM6DS3Class::end()':`
`libraries\Arduino_LSM6DS3\src\LSM6DS3.cpp:110:12: error: 'class TwoWire' has no member named 'end'_wire->end()';
```

This issue is only present when compiling for any ESP32 boards. When I switched the board to a Arduino board, it compiles.

Further investigate reveals, on both [Arduino ESP32's Wire.h](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h) and the [Arduino Standard Library](https://www.arduino.cc/en/Reference/Wire) use endTransmission(), not end().

A hotfix inside LSM6DS3.cpp that got my library to compile would be of the following:

```C++
void LSM6DS3Class::end()
{
  if (_spi != NULL) {
    _spi->end();
    digitalWrite(_csPin, LOW);
    pinMode(_csPin, INPUT);
  } else {
    writeRegister(LSM6DS3_CTRL2_G, 0x00);
    writeRegister(LSM6DS3_CTRL1_XL, 0x00);

    _wire->end(); // This does not compile for esp32. Somehow works for Arduino. Legacy code?
    // _wire->endTransmission();
  }
}
```
I'm wondering if the above code is all that is required to fix this issue.

cjinn, 2020-11-26:
Forgot to mention my library/environment versions:

- Arduino IDE v1.8.13
- Arduino ESP32 v1.0.4
- Arduino_LSM6DS3 v1.0.0

abrahmx, 2022-08-05:
> I noticed, when trying out this library for a ESP32 and I2C project, this library does not compile due to the following errors:
>
> ```
> libraries\Arduino_LSM6DS3\src\LSM6DS3.cpp: In member function [...]

Thank you for this, I'm using ESP8266 and I was getting this error as well. After changing this I could compile and upload. My settings are: Platformio(v2.5.1) on VS Code, Platform espressif8266, board = nodemcuv2, framework = Arduino.

Regards