arduino-libraries/Arduino_LSM6DS3 #41: readTemperature() Returns Nonsensical Values (Nano 33 IoT)

ryanohoro, 2023-12-07:
Running [SimpleTempSensor](https://github.com/arduino-libraries/Arduino_LSM6DS3/tree/master/examples/SimpleTempSensor) results in readTemperature() returning nonsensical values that change based on the temperature.

```
Temperature sensor sample rate = 52.00 Hz

Temperature reading in degrees C
T
123.12
126.44
126.00
127.25
128.56
```

The root cause is that the original part the Nano 33 IoT shipped with, the LSM6DS3 (Chip marked SH) is no longer manufactured, and other parts from this family are being shipped on these boards. Mine has a LSM6DSOX (Chip marked S4). These parts have different values for temperature sensitivity, which affects how the raw value is converted into degrees Celsius.

https://github.com/arduino-libraries/Arduino_LSM6DS3/blob/2640521a84dd3ee8c7894d1a5bdf48c4a2ff1f26/src/LSM6DS3.cpp#L162

[image]

[image]

Changing the formula results in accurate temperature values

`t = data[0] / 256.0 + 25;`

```
Temperature sensor sample rate = 52.00 Hz

Temperature reading in degrees C
T
26.95
26.83
26.79
26.95
26.95
```

Unfortunately, I'm not sure how to programmatically determine which temperature setting value is appropriate.

per1234, 2023-12-07:
Hi @ryanohoro. Thanks for your report.

> Mine has a LSM6DSOX

Arduino provides a different library named "Arduino_LSM6DSOX" specifically for that chip. You can install it from the Arduino Library Manager. Please give it a try.

https://www.arduino.cc/reference/en/libraries/arduino_lsm6dsox/

Even though the description mentions the Nano RP2040 Connect board specifically, the library metadata indicates it is compatible with boards of different architectures as well. I don't happen to have a Nano 33 IoT board with the LSM6DSOX chip, or a LSM6DSOX I can connect to one of my SAMD boards so I can test it out, but I think it is worth a try.

ryanohoro, 2023-12-07:
Thanks so much @per1234 ! I'll use the Arduino_LSM6DSOX library since it's more suitable for my Nano. Shame that this isn't documented on the Nano 33 IoT product pages, and that it's actually kind of hard to determine which chip you have, and therefor which library to use.

ryanohoro, 2023-12-07:
Update from tech support:

> While we strive to keep our documentation accurate, there have been instances where component changes are necessary due to various factors, such as supply chain disruptions or improvements in technology. In such cases, we aim to update our documentation as promptly as possible.
 
> Unfortunately, in this instance, the documentation on the IMU in the Nano 33 IoT product page seems outdated. We apologize for any inconvenience this has caused.
 
> We appreciate you identifying the correct library (Arduino_LSM6DSOX) for your specific board. We will investigate the issue further and update our documentation to reflect the actual IMU being used in the Nano 33 IoT (ABX00032).
 
> We are also working on a more comprehensive solution that will automatically detect the installed IMU and recommend the appropriate library. This will ensure that users have the best possible experience with their Arduino boards.
 
> Have a nice day!